dexterbg/Twizy-Virtual-BMS #1: Charger compatibility

dexterbg, 2017-10-20:
Blazej has reworked a handfull Twizys using the VirtualBMS and has found an issue: the current VirtualBMS implementation does only work with chargers up to generation 5 (charger label: "Ind …"). Generation 8 and newer do not work, generations 6 & 7 have not yet been tested.

The effect is the charger being stuck in the first protocol init step (F0-A2 in 597h) for both driving and charging. We've found no differences yet on the CAN side. We also tried some variations on the 3MW timing, but without success.

Please reply if you can share some insight or information on this problem or can help solving this issue.

Jekyll555, 2017-10-31:
As I did not find chargers around with Ind >5: 
Does this change in Charger implementation possibly coincident with the SEVCON (inverter) register locks (>06/2016)?

dexterbg, 2017-11-01:
No. Charger evolution timeline is roughly:
2011 - Ind 04 / PIE 296 054 684R -- works
2012 - Ind 05 / PIE 296 055 169R -- works
2013 - Ind 06 / PIE 296 052 118R -- unknown
2013 - Ind 08 / PIE 296 058 331R -- doesn't work
2014 - Ind 10 / PIE 296 050 463R -- doesn't work

klausz, 2017-11-01:
When talking to a Renault Dealer and looking what Power Supply has to be installed in case of a failure for a June 2012 Twizy (one of the first models) the following:
- the newest Power Supply can be installed. Means there is only one Power Supply Type for all Twizy (the earliest and the latest Models)
- when asking if there are additional parts needed, nothing was on the spare part/exchange list
- also they made no firmware update etc (they are not aware that there is a change). 
Min one failed power supply was exchanged not long ago (from an old Twizy) from them.
((When remembering correct, in the Twizy forum one person, who had exchanged the Power Supply himself, reported that the vehicle was not working)).

If everything above is correct (really?), 
- the Renault/Sevcon/LGChem firmware is working with all power supplies 
- whilst the Arduino Virtual BMS is not. 
- We currently dont know about the edriver BMS.

dexterbg, 2018-01-30:
The issue has been solved by Bernd with verification by Blazej.
The cause was a missing handshake in id155[1], the high nibble needs to go 9→A→9 along with the 3MW pulse.
→ Commit f45ee40ae2fb9b89ab5795ab270df76c209e506f

closed 2018-01-30 by dexterbg