dexterbg/Twizy-Virtual-BMS #2: Need help. Twizy charger issue

Kamsta99, 2019-10-06:
Hi, my twizy is stuck charging at max 5A, tried a lot off settings and a lot of BMS (Blazej BMS, KlausBMS,  SimpleBMS). Unfortunely im still not so advanced in coding on Arduino, but i changed setChargeCurrent() value to 10 A min in code to make sure but no success. Charging percentage are not updated, i need to switch off->on. Any sugestions ?

dexterbg, 2019-10-06:
That sounds pretty confused. You normally cannot simply switch the software without understanding how they each work and what they need.

SimpleBMS uses a voltage reading to derive SOC, and the SOC to derive the allowed charge current. So if the voltage reading isn't configured correctly or doesn't work, both SOC and charge current will likely also not work. BlazejBMS is tailored to a 4 "cell" lead acid configuration, and KlausBMS to a 16 cell LFP setup.

All example implementations do extensive logging via serial. My suggestions are:
a) describe your setup
b) provide the debug log

Kamsta99, 2019-10-06:
At first i want to thank you for reply, i done battery using Li-Ion batteries LEV40 14cellsx3. Everything except charging works fine (driving no errors).
[image]
I did case for batteries.
[image]
Also using oryginal sockets
[image]
No errors on LCD

Kamsta99, 2019-10-06:
Also i got these debugs when charging
[image]

dexterbg, 2019-10-06:
Nice images, but without any substantial info on your actual setup.

a) Do you use a real BMS or the Arduino as the BMS?
b) How do you read the battery sensors, which sensors do you have?
c) Your log only shows the VirtualBMS core debug output, where are the debug outputs of the BMS model chosen?
d) And which one is that now? And how is it configured / modified for your hardware?

Kamsta99, 2019-10-06:
a) Im using Arduino Nano as BMS
b) We did voltage divider from 60V ->4,5V like note in code also used vcore show same result like voltage meter
c) I got log only in idle state Vcore, soc and temp looking at code i should have more ... 
d) I had most success on SimpleBMS, there files, i changed battery voltage 
[SimpleBMS.zip](https://github.com/dexterbg/Twizy-Virtual-BMS/files/3694847/SimpleBMS.zip)

dexterbg, 2019-10-06:
OK, that gets us one step further. Now I need the actual full log output of a charge attempt.
The SimpleBMS log should show the measurements and data derived from the measurements once per second from bmsTicker().

Kamsta99, 2019-10-07:
There is everything i got via log
https://www.youtube.com/watch?v=UT8BhkOUG44&list=LLmkOltXakQK5IzsuK87Dk0A&index=2

dexterbg, 2019-10-07:
OK, there's a bug in the SimpleBMS bmsTicker function.
In SimpleBMS.ino, please change the ticker line 107 from…
```C
  if (twizy.inState(Off) && (clockCnt % 100 == 0)) {
```
…to…
```C
  if (!twizy.inState(Off) && (clockCnt % 100 == 0)) {
```
…then get another log.

dexterbg, 2019-10-07:
As the whole ticker did only run in state "Off", that may have been the cause for your issue with SimpleBMS as well.

But that bug is only present in SimpleBMS, not in BlazejBMS or KlausBMS, so let's see where you get now.

Kamsta99, 2019-10-07:
WOAH 32A charging working and percentage goes up. Grab my generous donate.

Kamsta99, 2019-10-07:
Nah i miss comment section in donate. I dont want to anonymous but Kamsta99 👍

closed 2019-10-07 by Kamsta99

dexterbg, 2019-10-07:
You're welcome. Please send images of the build process and final result if you'd like me to include your build as another showcase.

Kamsta99, 2019-10-07:
[image]
Trying to understand how arduino works 
https://dl3.pushbulletusercontent.com/iGrAU85naZRIOcvgQ866JFKJROkRwXkn/20190831_203701.jpg
also some hardware problems we cant get GO light at first

dexterbg, 2019-10-07:
Please collect everything as a ZIP and send me that by email for the inclusion. Thanks!

abdurauf16, 2020-08-17:
my twizy is off but stop and n are on

abdurauf16, 2020-08-19:
hello, I have a twizy on the ride, the light is off, the go light is on, stop can, or you will send a sketch for acid batteries for a renault twisi

abdurauf16, 2020-08-19:
hello, I have a twizy on the ride, the light is off, the go light is on, stop can, or you will send a sketch for acid batteries for a renault twizy please help me dwxterberg please help

Abdurauf1234, 2022-02-02:
pleas help how to do virtual bms

Abdurauf1234, 2022-02-03:
I bought a Renault Tweezy without BMS and for 4 years I can’t make a virtual BMS, I can make a video on how to make your first virtual BMS based on acid batteries, I would buy a Renault Twizy accumulator, but I need to bring it from Korea if it will cost about 2000 to bring it from Korea to our country $ 3000 and in our country on average they earn $ 200 help please make a virtual BMS from acid batteries, you can throw off the connection diagram of the components