dexterbg/Twizy-Virtual-BMS #3: Hi Dexter again i have problem with SimpleBMS ...

Kamsta99, 2020-02-09:
Hi friend, i need help a little bit, Twizy driving fine but my charger sometimes randomly shutdown ... solution is to just unplug and plug again, but this force to babysit your car, is there any possible solution via code ? Someting like loop to start charging again if SOC below value ? Im not scripter or coder sadly.
[SimpleBMS.zip](https://github.com/dexterbg/Twizy-Virtual-BMS/files/4178125/SimpleBMS.zip)

dexterbg, 2020-02-11:
Great to hear it still works.

There is most probably a simple solution to this, but we should first get a clue about the reason for the aborts. Can you provide a debug log (`TWIZY_DEBUG_LEVEL` set to 2) of a charge process including an abort event?

Do you use the original Renault charger? Many Twizy have issues with the first generation of these. Maybe your charger is about to die. Most issues result from the charger getting too hot, in some cases it could be traced down to a blocked/malfunctioning fan. Did you check if the fan works normally?

You also should open and check the connector from the charger to the battery, there have been reports of these corroding / burning / melting.

Kamsta99, 2020-02-11:
Glad to hear you :D
Im using original charger
[image]
Charger was dismantled and cleaned, fan was also cleaned some time ago.
 I will try to record this log today but its totaly random

dexterbg, 2020-02-11:
Yes, that's version 06 from 2013, IIRC still the first generation (second begins with Ind 08).
Fingers crossed it doesn't die now, you certainly won't get a replacement on warranty…

Kamsta99, 2020-02-11:
I cant reproduce problem now, today charging without shutdown as i say its random. Did you have idea how to fix this by extra bits of code ? Code seems less expensive than charger

Kamsta99, 2020-03-11:
Hi after a longer while, i tried debug mode 2 but using this make my car unable to drive or charge, after dump everything crash (like after 10 seconds after run). My charger from 2013 likely died some days ago, i installed another which i bought from internet, charger from 2016, everything working well but still finish charging before reaching 100%, any ideas ?

dexterbg, 2020-03-15:
Well, if you bought a used charger, you cannot know if that doesn't have the same issue already…
Debug mode 2 normally works, no idea about the crash. Regarding the charge stops, another (maybe better) option would be a separate CAN log.

Kamsta99, 2020-03-15:
Debug mode 2 make my twizy not able to charge or drive, when dump appear via bluetooth GO light disappear and if trying to drive km/h also. Charger (new one) working well, sometimes shutdown near 80%-90%. Separate can log how to do that ?

dexterbg, 2020-03-15:
Use some separate CAN logging device. Needs to be capable of 500 kbit CAN logging. You can use my https://github.com/dexterbg/CanLogger, but take care to turn on compiler speed optimization, comment out the logic inputs, and/or use a faster Arduino board.

Kamsta99, 2020-04-25:
I have no clue how to do that, tried some things, but no success. I started to doing code solution.
`    }
    if (twizy.isPluggedIn() == 1) {
      if (twizy.inState(Off) && (soc <= 98)) {
        Serial.print(F("bmsEnterState: forcing charging "));
        twizy.enterState(StartCharge);
      }
    }`

Kamsta99, 2020-04-25:
Changed code a bit got 3MW working, but still keeps staing in off state
[image]
[image]

dexterbg, 2020-04-26:
Are you trying to (re-)start the charge from the BMS? That will not work. The BMS doesn't start the charge, the charger does. The BMS can tell the charger to reduce current or stop, but not to start.

My idea was there is possibly something causing a stop charge command, but you should see that in the BMS log. If you don't, the charger stops by itself, and then the BMS cannot do anything about that.

There's just one slight possibility that may also cause the charger to abort: if your Arduino has a timing issue, i.e. cannot sustain exact 10 ms intervals for the communication. See config #define TWIZY_CAN_CLOCK_US. Try reducing and increasing this in small steps around the default of 10000.

Kamsta99, 2020-04-26:
Problem happen unfortunnelly on 4 different complete arduino boards

dexterbg, 2020-04-27:
I still think you need to capture a debug log and/or CAN log of an abort event first to get a hint about the source of the issue.

Kamsta99, 2020-06-08:
https://streamable.com/ivw80g
This is how this looks on twizy debugmode 2
Still no can logging sadly

Abdurauf1234, 2022-02-02:
Hello l have Renault twizy but not bms how to do simple l make simple bms but don.t working

closed 2022-02-03 by dexterbg

Abdurauf1234, 2022-02-03:
how to make a virtual bms for tweezy acid or gel I need a virtual bms for the answer please

Abdurauf1234, 2022-02-03:
don.t close please

Abdurauf1234, 2022-02-03:
you have instagram or facebook please help

Abdurauf1234, 2022-02-03:
I bought a Renault Tweezy without BMS and for 4 years I can’t make a virtual BMS, I can make a video on how to make your first virtual BMS based on acid batteries, I would buy a Renault Twizy accumulator, but I need to bring it from Korea if it will cost about 2000 to bring it from Korea to our country $ 3000 and in our country on average they earn $ 200 help please make a virtual BMS from acid batteries, you can throw off the connection diagram of the components

Abdurauf1234, 2022-02-03:
please help

Abdurauf1234, 2022-02-03:
my email abduraufvaliev@gmail.com

Abdurauf1234, 2022-02-04:
dexterberg