dexterbg/Twizy-Virtual-BMS #4: Start signal fault

Rox50055, 2020-03-31:
Hello!
Thanks to your code we got super far with our twizy! most fault codes are gone, but some annoying ones stick. 

The first one is like the title: Start signal Fault P0412.
Along that code we also got: Power supply relay Fault P0218. We got a relay connected to the Sevcon relay wires on the plug coming from the Twizy.
The red STOP light is on and and the battery logo with ! on it, no ERROR though.

For now we are just connecting the bare minimum to test if it still works but to no success yet.
I was wondering if you might know a fox for this issue?

Hope to hear from you soon!
Greetings Kenneth

dexterbg, 2020-04-04:
Kenneth,

the fault sounds like you only connected only one of the SEVCON supply relays.

I've attached a draft of the Uno example extended by the Yazaki plug and relays. Check your setup against this.

Regards,
Michael

[image]

Rox50055, 2020-04-11:
Thanks for the reply!
Making these connections solved the No start signal problem. But sadly the twizy still doesnt work.

We now got the fault P0216/DF181 "Power Relay Charger" and it wont activate the relays to power the sevcon or or other components. When I have my can reader attached with live data I can see the states of all relays and by default it's Open and when I manually power it, it switches to Closed. Meaning that the feedback works. 

My thought is that the charger is not giving out the OK signal for it to work or not getting all the proper information. It is a Twizay and charger from 2012, so there shouldn't be any software locks.

I hope that you might know a solution to this!
Thanks in advance

Best regards, 
Kenneth

dexterbg, 2020-04-13:
The charger power relay is controlled directly by the charger, it's normally activated during the startup handshake with the BMS (not sure at which step, but I'd guess immediately after the initial communication). Compare the handshake to the protocol document. If the handshake looks normal, there must be some electrical fault. Check all wiring, connections and fuses involved.

Tudy1983, 2020-05-04:
Can have the code please compiled I'm trying to figure out how this is done I'm new in this project I'm still learning like this I can make my own battery  thanks

Rox50055, 2020-05-04:
Hey. 
Sadly we have determent that the charger is broken (it's the oldest model) and that's why it doesn't work. The new problem is that we can't find a new, old one that will work with the VBMS you wrote according to what I read on your pages. (The 2016+ models)

We also tried to control  the sevcon direcly with no success since it's also expects a Clear from over the canbus. 

I do not have the equipment or know how to decode canbus signals, nor do we have a working twizy. Maybe you know a way to remove the charger completely out the system? 

Hope to hear from you soon again! 
If this way doesn't work than we probably have to just get rid of it all and get a universal ESC/Motor controller.

Best regards, 
Kenneth

Tudy1983, 2020-05-04:
Wath I was mean is the code for arduino I'm tryna o se the difference between them I'm trying to build a virtual bms an I think that somewhere I made a mistake

dexterbg, 2020-05-09:
Kenneth,

> Sadly we have determent that the charger is broken (it's the oldest model) and that's why it doesn't work. The new problem is that we can't find a new, old one that will work with the VBMS you wrote according to what I read on your pages. (The 2016+ models)

That is not correct. All charger models are supported.

> We also tried to control the sevcon direcly with no success

You cannot operate the Twizy without the charger, as the charger runs a special Renault firmware that works as the central power up controller for the whole system. There's a chance to implement a virtual charger as well, based on the information provided here. But don't expect that to be done by me.

Regards,
Michael

xury77, 2021-07-15:
I have same problem. Did you resolved it? 
I also have a question about 3MW relay. It should be activated (shorted C with NO) by LOW gpio3 state or HIGH? I bought one witch is activated by gnd signal. Its correct?

dexterbg, 2021-07-16:
@xury77 The code assumes HIGH (1) to activate the relay. You can try exchanging 0/1 on all ``digitalWrite(TWIZY_3MW_CONTROL_PIN, …)`` calls if your relay needs it the other way around. You'll probably also need an init to 1 after ``pinMode(TWIZY_3MW_CONTROL_PIN, OUTPUT);``.

Regarding the fault issue, maybe @Rox50055 can provide some info on his advances.

xury77, 2021-07-16:
Hi Michael. Could you take a look at this diagram. Is the voltage divider is correct? Or I have to swap resistors?
[image]

dexterbg, 2021-07-17:
Looks correct to me. Be aware the voltage divider is a rather simple monitoring approach, it didn't work very well for Klaus. Readings will be off whenever the BMS activates a cell bypass. There are better solutions based on Opamps or dedicated ICs like the LTC6803. Best option is to use a BMS that can give you the cell voltages via some communication interface.

kosemrt, 2021-10-04:
> Kenneth,
> 
> the fault sounds like you only connected only one of the SEVCON supply relays.
> 
> I've attached a draft of the Uno example extended by the Yazaki plug and relays. Check your setup against this.
> 
> Regards, Michael
> 
> [image]

Hello could you share this relay' fritzing file please?

closed 2022-02-03 by dexterbg

abdurauf12345666, 2022-02-03:
I bought a Renault Tweezy without BMS and for 4 years I can’t make a virtual BMS, I can make a video on how to make your first virtual BMS based on acid batteries, I would buy a Renault Twizy accumulator, but I need to bring it from Korea if it will cost about 2000 to bring it from Korea to our country $ 3000 and in our country on average they earn $ 200 help please make a virtual BMS from acid batteries, you can throw off the connection diagram of the components

abdurauf12345666, 2022-02-03:
please help me