dexterbg/Twizy-Virtual-BMS #5: Battery charge level

bakrimo, 2020-04-18:
Hi friend
I did simple bms (lead acid battery)
Problem that the charge level not showing nothing
Any suggestions what I did wrong
Thanks

dexterbg, 2020-04-18:
https://www.youtube.com/watch?v=x40XnMGI75M

bakrimo, 2020-04-19:
Thank you for your replay
The battery level on dashboard it shows 0v charge like no battery

dexterbg, 2020-04-20:
That's what you initially wrote. What about details like hardware setup, software configuration & serial debug log content?

bakrimo, 2020-04-21:
What find out that the problem with hardware
Mcp2515_can bus that I have is 8mhz
Instead of 16mhz
Is there anyway to let it work with 8mhz

dexterbg, 2020-04-21:
Looking into the configuration file may help…

https://github.com/dexterbg/Twizy-Virtual-BMS/blob/master/examples/SimpleBMS/TwizyVirtualBMS_config.h#L33

But earlier attempts with 8 MHz failed, that's why I recommend 16 MHz. You normally can exchange the crystal on an 8 MHz module.

bakrimo, 2020-04-27:
I replaced the 8mhz crystal with 16mhz but still same problem is there anyway to get shem of Arduino nano and mcp2515 connection

dexterbg, 2020-04-27:
I don't understand you. If there was a basic issue with your MCP2515 connection, you couldn't even switch into drive mode.

bakrimo, 2020-04-28:
Exactly

dexterbg, 2020-04-28:
A quick duck reveals this NiRen/Nano guide: https://circuitdigest.com/microcontroller-projects/arduino-can-tutorial-interfacing-mcp2515-can-bus-module-with-arduino

bakrimo, 2020-05-03:
You are right was hardware problem, "Nano cheap quality built". 
Now it's showing battery level on dashboard full,  but I don't receive command from Pin9 relay sevcon supply to turn relay on
Any suggestions!

bakrimo, 2020-05-08:
What about "ecu ok" should connect to 12v bms by 10kohm resistant or 470 ohm

dexterbg, 2020-05-09:
Check out the example: https://github.com/dexterbg/Twizy-Virtual-BMS/blob/master/extras/Arduino-Example-Yazaki-Relays.pdf

bakrimo, 2020-05-10:
First thank for your reply
I did this example, when I turn the ignition On it shows battery level on dashboard everything OK and N flashing. When I turn key to start position nothing happens, and I tested relay K1 sevcon supply doesn't click (doesn't receive order to work)

bakrimo, 2020-05-14:
[image]
That is the charger I have on twizy, you think simplebms can work with it

dexterbg, 2020-05-16:
There are no known charger incompatibilities.

bakrimo, 2020-05-16:
Anyway when I plug 🔌 220v the charger wouldn't give 60v to traction battery only 13.5v to the 12v battery. Does it need signal from can_bus? 
Or the charger damaged

dexterbg, 2020-05-16:
Yes of course it needs the CAN handshake with the BMS, that's what this project & repository is all about. See protocol documentation in the extras folder. I can only help you with the software if you provide debug details.

bakrimo, 2020-05-18:
[image]
[image]
[image]
[image]
Hopefully will help

bakrimo, 2020-05-20:
Do we  have to code the BMS battery to the rest of the vehicle…charger and Inverter doesn’t recognize the BMS

dexterbg, 2020-05-21:
Taking photos of the serial log is really weird. You can simply copy the serial log output.

But regarding the log, that looks perfectly normal & correct. So there doesn't seem to be a CAN or software issue, except if you would have configured the wrong output port for the 3MW relay. But you would already have noticed the 3MW relay doesn't work, wouldn't you?

If 3MW is OK, you should check if the CAN transmissions of the VirtualBMS really get to the bus. And you should check all charger and inverter wirings to the BMS, probably also check the relays.

bakrimo, 2020-05-22:
Every things looks good with the wires.
But when the key on M position N keep flashing is it normal? 
And when I turn key to start position like anything at all happens

bakrimo, 2020-05-28:
Actually 3MW relay once I turned the ignition to position ON it turned off. And the and the STOP LIGHT WITH 3 BEEPS

closed 2022-02-03 by dexterbg