dexterbg/Twizy-Virtual-BMS #8: Hi Dexter after long time

Kamsta99, 2021-07-28:
I did Arduino project based on SimpleBMS, with very precise voltage meter (0.03 voltage accuracy), because used way is way to unprecise to use (0.3-1v accuracy) based on voltage delivered to Arduino and temperature. I expanded project with multiple security option like 3 temperature sensors, various temperature alerts and limitations, with use of ads1015 to have precise voltage based on project: https://www.youtube.com/watch?v=10QtHh6UNBs, merged toggeder. I want will send you soon my project to use :)
[image]
[image]

I want to ask you, did its possible to control range showed in Twizy by battery ?! If yes tell me how pls :D

dexterbg, 2021-07-29:
Nice :-)

> I want to ask you, did its possible to control range showed in Twizy by battery ?!

Not to my knowledge. AFAIR the Twizy display unit calculates the range by itself from the distance driven and the SOC consumed.

Kamsta99, 2021-07-29:
That sad, i found that here but i dont know how interfere with that
[image]

Kamsta99, 2021-07-29:
[SimpleBMS.zip](https://github.com/dexterbg/Twizy-Virtual-BMS/files/6900315/SimpleBMS.zip)

Kamsta99, 2021-07-29:
> 
> 
> Nice :-)
> 
> > I want to ask you, did its possible to control range showed in Twizy by battery ?!
> 
> Not to my knowledge. AFAIR the Twizy display unit calculates the range by itself from the distance driven and the SOC consumed.

Problem is i drived on single charge 91 km but on display show max 30 km range

mucahitdemir, 2022-01-05:
Hi! I think mathematical expressions for displaying _taken range_ may lead to wrong results.
In addition, can you debug this calculation by _stepping into_ function and _stepping over_ till the end?
You can track the problems on STM32 IDE or System Workbench IDE through debug options.
Best,
MD

closed 2022-02-03 by dexterbg