dexterbg/Twizy-Virtual-BMS #9: Twizy sevcon voltage limit

Kamsta99, 2021-08-03:
Source: https://github.com/dexterbg/Twizy-Virtual-BMS/blob/master/extras/SEVCON-Configuration.md
obraz
I tried doing this via Ovms, no success still cutsoff after 39 V

Patient:
Twizy, 2012 ,custom 8.7 kwh battery,

What i tried already:
Setting Twizy same as for applying tuning (no switching to GO)
In shell:

    enable
    xrt cfg pre
    xrt cfg write 2c02 01 592
    xrt cfg write 2c02 02 560
    says --> OK, value edited is same as we set
    But still cutsoff, with STOP light and SEVCON low voltage error

dexterbg, 2021-08-04:
I'm afraid you're the first to actually try running the Twizy on such a low voltage. It's possible the Renault SEVCON firmware has some fixed internal limits to protect their battery, so the SDOs do not work as expected.

Another option is there is another SDO you need to change. Check out the "master dictionary" from the DVT software, maybe you can find some matching description.

Kamsta99, 2021-08-07:
Thanks for reply, where i can get DVT software for not 500$ and stuff
needed for that also?

Kamsta99, 2021-08-07:
Did we able to reprogram (unlock speed for example or torque) sevcon after
july 2016 with DVT software and usb dongle ?

dexterbg, 2021-08-07:
a) Search the web, b) no.

closed 2022-02-03 by dexterbg